ImperialSpaceSociety/picotracker-Lora #101: [MAJOR] when new v1.0.2 application is made with US915 settings, no packets received in EU

MedadRufus, 2021-05-08:
On ttnv3, when a new  v1.0.2 application is made with US915 settings, no packets received in EU. Note that the same application can be created with EU868 settings and receive the data.

Changing region in teh interface to EU868 fails with the following error:
```
invalid value of field mac_state.current_parameters.rx2_data_rate_index

{
  "code": 3,
  "message": "error:pkg/networkserver:field_value (invalid value of field `mac_state.current_parameters.rx2_data_rate_index`)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/networkserver",
      "name": "field_value",
      "message_format": "invalid value of field `{field}`",
      "attributes": {
        "field": "mac_state.current_parameters.rx2_data_rate_index"
      },
      "correlation_id": "fc1143302d2e4fb7aac6a8d07da3afe4",
      "code": 3
    }
  ],
  "request_details": {
    "url": "/ns/applications/icss-lora-tracker/devices/icspace26-ttnv3-abp-us-v1-0-2",
    "method": "put",
    "stack_component": "ns"
  }
}
```

MedadRufus, 2021-05-08:
Only solution may be to switch ABP keys for each region, and make a seperate ABP application for each region.

MedadRufus, 2021-05-08:
Safest solution is to create ABP keys for each region. Each registration must have reset fcount enabled AT REGISTRATION. It apparently cannot be changed later.

Then when each region is selected, the keys should be changed along with it. Create look up table {region, network key, appsession key, id}

MedadRufus, 2021-05-08:
Changing region after creating the application does not work, although the interface indicates it.

MedadRufus, 2021-05-08:
This is the case for lorawan v1.0.3 as well

MedadRufus, 2021-07-10:
Just make new keys for each region

pimente, 2021-09-27:
Hello @MedadRufus
Congratulations for the success you have had so far. 
This is quite impressing to have achieved circumnavigation with Lora only!

I am currently working on a similar project and have already achieved a successful flight in Europe with TTN3 and a Lora-E5 based tracker.
I am now trying to make the code work all over the world. 
From the issues you have raised, it is not clear to me if we need to set up new ABP keys for each region. 
This issue (the latest) seems to suggest that a key is needed for each region while issues [71](https://github.com/ImperialSpaceSociety/picotracker-Lora/issues/71) and [92](https://github.com/ImperialSpaceSociety/picotracker-Lora/issues/92) tend to suggest the opposite.

What final conclusion have you reached regarding this topic?

Your help is very much appreciated.

Cheers

MedadRufus, 2021-09-27:
My conclusion is that separate ABP keys are needed for each region. On the Things Network V3 console, I register one ABP device for each region, and dynamically change the ABP keys(DevAddr, FNwkSIntKey, SNwkSIntKey, NwkSEncKey, AppSKey) depending on each region.

In TTNv2, it was possible register a single ABP end node, and it would work in all regions(US915, EU868, AS923). This has been tested in flight. There was no requirement to specify which region the device was registered to in the interface.

However, on TTNv3, when registering the device, its required to specify which region parameters the end node uses. It appears to me that the set of ABP credentials generated for each region will only work on those region frequencies. I have tested this by:
1. registering an end node with US915 region
2. transmitting on EU868 frequencies and receiving on an EU868 gateway.

I get no packets on the TTNv3 console. So I conclude that an end node registered to a region will not be received in any other region. Hence, I register a new device to each region(US, EU, AU, RU etc) and then switch the credentials on the end-node dynamically.

I have now abandoned this repo in favour of a fork off the Semtech reference code which can be viewed here. https://github.com/ImperialSpaceSociety/LoRaMac-node. I needed to use the Non-volatile memory to store all the keys for an OTAA activation, to avoid repeated joins after resets and this later version of the stack has this functionality.  I am also getting the code to switch between the Things Network(ABP only) and The Helium Network(OTAA only) every 3 transmissions, to increase my coverage.

Below is a list of all my devices for each region.
[image]

pimente, 2021-09-28:
Thank you for the very thorough answer!

I am using the [mbed-os](https://os.mbed.com/mbed-os/) framework with [mbed studio IDE](https://os.mbed.com/studio/). A lot of different hardware is supported by [stm32customtargets](https://github.com/ARMmbed/stm32customtargets) and the example [mbed-os-example-lorawan](https://github.com/ARMmbed/mbed-os-example-lorawan) seemed much lighter than the Semtech LoRaWAN example which has a lot of boilerplate in my opinion. Not sure if mbed provides the same functionalities than the original Semtech code thought.

I did not know about the The Helium Network. It seems to have a great coverage. I now understand your need and reflexion around OTAA...
With your alternating transmission scheme, did you get a good sense of the Helium versus TTN coverage?
Does it really help to add the Helium Network?
If so, I'll give it a try.

From your screenshot, it seems that icspace26 is doing great with a flight longer than 110 days!? I can't wait for the flight report if you share it 😄 

Cheers

MedadRufus, 2021-09-28:
>  With your alternating transmission scheme, did you get a good sense of the Helium versus TTN coverage?
I have yet to launch this tracker, with alternating helium/TTN transmissions. So I don't know. One think I have seen is the helium coverage is much better than TTN. Outside my house in London, UK, I immediately managed to successfully join the Helium network in a few minutes, using gateways nearby. On the Things Network, I have almost never gotten public gateways to pickup packets.

pimente, 2021-09-28:
> well ICSPACE26 is yet to be launched, only testing for 110 days 😄

Finger crossed! Do you plan to launch soon?

> I have looked very closely at the Semtech stack, and the only unnecessary part in there is the LoRaWAN Class B support, which I will never be using. Its not a big problem for me, as the compiled code size is 182 kb, less than the max of 192 kb for the STM32l072 processor I use. Ram is also under control, at around 11 kb, less than max of 20kb. I have chosen to use this stack because I know its well tested by the developers. I am less trusting of other ports of this code.

I agree that reliability is the most important criteria when using others' code for these flights. I've seen that your new fork introduces some UTs. That should help.

For the Helium Network, do you need multiple credentials for each region?
If not, and considering the coverage, I might move to the Helium Network instead of configuring TTN for each region...
Mbed implementation is lorawan 1.0.2. I see that you use version 1.0.3. Is it mandatory to connect to the Helium Network?

MedadRufus, 2021-09-28:
> For the Helium Network, do you need multiple credentials for each region?

Good question. I don't have the answer for this. To be safe, I have created registered 3 different end node(with unique keys) for the US, EU and CN. When registering a Helium network end node, I don't specify which region it will be used in. So one registered device might work in all regions around the world(US, EU, CN). However, I cannot test this as I don't have Helium Gateways for these 3 regions. So to be safe, I created 3.

MedadRufus, 2021-09-28:
Unlike ABP devices, OTAA devices require a join request to be made, which includes a downlink from gateway to tracker. This is not reliable I discovered in previous flights. The end node may not receive the downlink from gateway, and cannot transmit data at all.  So I want it to do only one join request in its lifetime. To do that, I save the keys and channel lists to non volatile memory so that after a reset, it does not have to rejoin.

MedadRufus, 2021-09-28:
> I agree that reliability is the most important criteria when using others' code for these flights. I've seen that your new fork introduces some UTs. That should help.

I need to document how to run the unittests. It uses CMAKE to build and CppUTest framework. I normally run it in VS code which has the test adaptor which makes running it very easy.

MedadRufus, 2021-09-28:
>  Mbed implementation is lorawan 1.0.2. I see that you use version 1.0.3. Is it mandatory to connect to the Helium Network?

I don't have any information on which version to use. LoRaWAN version 1.0.3 is what I have used on the Things Network V2, and also works on TTNv3. And in testing, also works on the helium network.

MedadRufus, 2021-09-28:
> Finger crossed! Do you plan to launch soon?

Hoping to launch by the year end

pimente, 2021-09-30:
> One thing I have seen is the helium coverage is much better than TTN. Outside my house in London, UK, I immediately managed to successfully join the Helium network in a few minutes, using gateways nearby. On the Things Network, I have almost never gotten public gateways to pickup packets.

Following our discussion, I gave a try to the Helium Network and made a few tests here in Paris. I can confirm your observations. I successfully join in a few minutes using multiple gateways nearby. On TTN, I had to stick my device on a fishing rod and wait hours to get an uplink with ABP.

> Unlike ABP devices, OTAA devices require a join request to be made, which includes a downlink from gateway to tracker. This is not reliable I discovered in previous flights. The end node may not [...]

This makes a lot of sense. I have also observed that I often need multiple Join Requests before the node actually receives the Join Accept. However, if I think about your other comment:

> To be safe, I have created registered 3 different end node(with unique keys) for the US, EU and CN. When registering a Helium network end node, I don't specify which region it will be used in. So [...]

I understand that when you change region you will change credentials and make a new Join Request with the new region's credentials. Is that correct? Will you you try to Join with the new credentials if you can't transmit with the previous ones? I see a potential failure scenario here. When you change region:
- if you make a new Join Request with the new region's credentials, you risk of never receiving the Join Accept
- if you try to stick to the saved credentials, you need to confirm that the transmissions still work so you also need to receive downlinks.

Unless I am missing something here, in both scenarios, you are depending on a downlink message, which sounds risky.
If so, I think that by trying to be safer and having 3 credentials we might take more risk than if one credential was actually enough. With one credential and a unique Join request at the launch site, we are not depending on downlinks for the full flight.
If you agree with this, I think we should try to figure out if one credential is enough maybe by contacting someone outside Europe in the pico group and try with European credentials? That would be simpler and safer if is works.
Looking forward to your view on this.

MedadRufus, 2021-09-30:
The semtech LoRaMac-node implementation saves the Channel Frequency list(CF list) to eeprom, along with the region's keys(DevAddr etc). I am a bit reluctant to modify the what parameters they save to EEPROM. I simply allocated 3 blocks of EEPROM, one block for each region, and make the tracker save all the keys, channel lists, frame count to the block that corresponds to that region.

pimente, 2021-09-30:
Very thoughtful!
Do you know if the CF list if always the same for each region? If so, we could only change the CF list and keep the same set of OTAA credentials.

MedadRufus, 2021-09-30:
I suspect the CF list will be different for each region. But I also suspect you can get the tracker to ignore the CF list, and only transmit on the default channels for that region(the EU has 3 default channels for example)

MedadRufus, 2021-09-30:
But i have a feeling that we can use the same OTAA keys in all regions. This has to be tested, as you say, with somebody in another region of the world(e.g. US). I will try to test with the  [Virtual Lorawan device](https://github.com/helium/virtual-lorawan-device). I have gotten the virtual device to work on TTNv2 but not yet on helium network

pimente, 2021-09-30:
That would do the trick! I've once forced a tracker to transmit only on the first channel, to have it work with a Single Channel Gateway. I had to watch out manually for duty cycle and it was not really lorawan compliant anymore. 
But maybe just ignoring the CF list and transmitting on the default channels (not just the first one) would not break lorawan compliancy. Good luck on the virtual device. It will be useful in all scenarios even to test a single set of OTAA keys for all regions I suspect.

pimente, 2021-09-30:
One side question: I am gearing up on the Helium Network and I am experiencing difficulties to always receive Join Answer. By looking at the Helium dashboard, I see a delay between the Join Request and Join Answer often around 2000ms which does not align with the default EU868_JOIN_ACCEPT_DELAY1 which is set to 5000ms. Did you have to tweak this parameter for the Helium Network?

MedadRufus, 2021-09-30:
I have not tweaked any parameters for the join request(or any parameters at all). I have the same behavior on The Things Network V2 and V3 and the helium network during a join.

MedadRufus, 2021-09-30:
The LoRaMac-node implementation uses a Join accept delay 1 of 5000 ms, as seen in the code: https://github.com/ImperialSpaceSociety/LoRaMac-node/blob/4027ac1ab855e449acf87140728d7d4d30bd71b3/src/mac/region/RegionCommon.h#L61-L64

pimente, 2021-09-30:
Yes, same as the mbed implementation (which uses the Semtech one). My issue must come from something else. I will investigate further. Thks! Let me know your progress with the Virtual Lorawan device. I am very much interested and will try to look into our idea of not updating the CF list once I've solved my problems with Join Request/Accept reliability.

MedadRufus, 2021-10-01:
That's exactly what I was testing earlier. I can get it to work on the Things Network, doing a join there, then extracting the dev_addr, nwk_skey, app_skey. I did the join with the virtual device. Then I key in dev_addr, nwk_skey, app_skey into my code and operate as an ABP device. This works perfectly. The network thinks the device is the old OTAA device.

However, I cannot replicate it with the helium network.

MedadRufus, 2021-10-01:
For some reason, on the Helium Network, I hard code in the dev_addr, nwk_skey, app_skey into tracker firmware, and operate it as an ABP device. However, I see no packets on helium console. I am not yet sure why. Works fine on The Things Network but not Helium network.

pimente, 2021-10-01:
I see. I've done that with a real node too so I have just experienced the same issue. Do you think it could come from frame counters that we must also transfer or other settings ? I will dig in into what is modified in the LoraMac state by the Join procedure. The meb implementation is quite straightforward and uses the underlying Semtech one so that should related to your code as well. I extracted dev_addr, nwk_skey, app_skey from LoRaMac::handle_join_accept_frame() here: https://github.com/ARMmbed/mbed-os/blob/master/connectivity/lorawan/lorastack/mac/LoRaMac.cpp#L181

pimente, 2021-10-01:
I think we are quite close to a hacky but elegant solution 😄

MedadRufus, 2021-10-01:
I am not sure its the frame count thats an issue. In my code, I use random frame counts with each transmission(when using ABP). If the frame count does not increment, the server should ignore the message. However, since I use a random frame count with each tx, there is a 50 % chance that the frame count is larger than the previous. But still, I don't see packets on helium console.

To avoid all doubt, i will next extract the dev_addr, nwk_skey, app_skey and **Frame count** and use it as an ABP device.

The random frame count implementation is a workaround for some other issue.

pimente, 2021-10-01:
Ok, I think this rules out frame count as an issue. Indeed, now that you say it, I stopped my device just after the 1st link after joining with OTAA so frame count must have been kept low. Then with ABP, no uplink were received by Helium but I send a few of them.

MedadRufus, 2021-10-01:
I would indeed rule out the frame count as an issue based on your tests and mine. I am not sure what other information is needed to fool the helium network into thinking the device is an OTAA device(not ABP device).

MedadRufus, 2021-10-01:
Update. I made a mistake. The OTAA/ABP hack does not work on the Things Network either. I think I know why.

pimente, 2021-10-01:
You know why it does not work on TTN specifically or on both networks? Do you have an idea to fix? I can make some tests on my end tomorrow if that can help.

MedadRufus, 2021-10-01:
I don't know yet why it does not work on both networks. What I know now it definitely does not work. I thought it was an endianness issue when hardcoding in the dev_eui but it does not seem to be the case.

pimente, 2021-10-01:
Ha ha! I exactly though the same but looking at how the dev_addr is read from the rx buffer. I tried inverting only the dev_addr without success and planning to try tomorrow by inverting all credentials... I vaguely remember that the lora packet is somehow inverted but have to look in the specs to really understand.

MedadRufus, 2021-10-01:
The Things network console shows the dev_addr, nwk_skey, app_skey even for _OTAA_ devices. I hard code them and try to join as an ABP device, but it does not work. No messages appear on console. Endianess should be correct. I copied the keys direct from the console. So it seems to me that dev_addr, nwk_skey, app_skey and frame count is not sufficient to trick the LoRaWAN network that its an OTAA device. Not sure why.

[image]

MedadRufus, 2021-10-01:
OK I have finally figured out whats missing. The device must also store the AppEUI.

So to trick the network into thinking that an OTAA device is connected, the following pieces of information is needed(can be hardcoded):
1.DevAddr
2. AppEUI
3. NwkSKey
4. AppSKey
The Frame Count must also only increment(no decrement).

I can now get it working on The Things Network, registering as OTAA and hardcoding the DevAddr, AppEUI, NwkSKey, AppSKey and consistently incrementing fcount.

I got the clue from the LoRaWAN 1.0.3 spec. 
[image]

MedadRufus, 2021-10-01:
@pimente I am pleased to say that the helium network will work around the world with a single set of credentials. This is how I tested it.
1. I used a virtual lorawan device "transmitting" on US915 frequencies to complete a join request. Then I extract the `DevAddr`, `AppEUI`, `NwkSKey` and `AppSKey`. 
2. Then I hard code the above keys into the code for an ABP device on my physical tracker. Then I take it outside my house in London, UK, transmitting on European frequencies and immediately packets start appearing on my console. Success!

So since it worked across 2 regions, it will likely work around the world.

pimente, 2021-10-01:
@MedadRufus Double win! This is a breakthrough for circumnavigation flight! Congrats! Great guess work. I will try to reproduce tomorrow from Paris, FR. Do you think there will be some king of expiration on theses credentials?

MedadRufus, 2021-10-01:
Good question. I don't know if there is an expiry, when the credentials become invalid.
I must also point out, the device must be able to keep track of the frame count, so that its always incrementing.

MedadRufus, 2021-10-01:
For reference, I link the instructions for running the virtual lorawan device for the helium network here: https://github.com/ImperialSpaceSociety/LoRaMac-node/issues/113#issuecomment-932142179

pimente, 2021-10-02:
I have not been able to reproduce yet. In fact, I realised that I was already setting the AppEUI even with ABP. But I don't see it being used anywhere in the code with ABP...
This paragraph 6. 1 of LoRaWAN 1.0.3 spec is strange indeed, especially if you compare it to paragraph 6.3 which states that skeys are stored directly instead of DevEUI, **AppEUI**, and AppKey:
[image]

Quick question: Can you confirm that the DevAddr and AppEUI you generated with the virtual lorawan device and you use for ABP are the ones you see in the Helium console?

AppEUI in DEVICE DETAILS:

[image]

And DevAddr in the General tab of an Uplink event:

[image]

 Thanks!

MedadRufus, 2021-10-02:
@pimente I know the problem. Its the endianness of the dev_address. Try devaddr = 0x480002AC

MedadRufus, 2021-10-02:
I know that the helium network uses dev_add values with 0x48xxxxxx. This is documented at the bottom of the docs: https://docs.helium.com/lorawan-on-helium/

MedadRufus, 2021-10-02:
One more thing to note: every time you do a join, a new dev_addr, NwkSKey, AppSKey are generated. The old ones may become invalid.

pimente, 2021-10-02:
The screenshot I have shared is a successful uplink just after the OTAA. I was suspecting from the Lora implementation's code that endianness was mixed up in the Helium console but the link you've shared just proves it. It is a bug in their interface. 
The problem is that I have already tried with devaddr = 0x480002AC without success. I feel am missing something from your last discovery. Looking at the code, I don't see where AppEUI is being used with ABP. Have you understood that?

pimente, 2021-10-02:
I'll try again with devaddr = 0x480002AC just to make sure.

MedadRufus, 2021-10-02:
I have not looked closely at the implementation difference between OTAA /ABP in the Semtech reference code. However, I am sure the AppEUI is required for OTAA. So even when using ABP, i think the AppEui must be used somewhere.

MedadRufus, 2021-10-02:
Ill try will incorrect AppEUI and see if it will work.

MedadRufus, 2021-10-02:
Correction from me. I switched around a few bytes of my hard coded AppEUI, and it still works. This proves to me that actually, the AppEUI is not necessary. I think you caught it @pimente

MedadRufus, 2021-10-02:
The only keys that have to be correct I think are back to the `DevAddr`, `AppEUI` and `NwkSKey`. When I was testing yesterday, I keyed in the correct AppEUI, and immediately, it started working. It may have been some other cause that made it to work.

MedadRufus, 2021-10-02:
I would recommend trying with the Things Network first, as it is a lot more transparent. It shows the `DevAddr`(with correct endian), `NwkSKey` and `AppSKey` on the console. Then if it works, the move on to the Helium Network. It helps to have a gateway at hand to test with.

pimente, 2021-10-02:
This works for me as well! I suspect problem was coming from ADR but I still have to investigate this. 
Switching to ABP, default DR was DR_5 which I though was not low enough to be received from the nearest gateway so I disabled ADR and tried different DR down to DR_0 without success.
Following our discussion, I joined again with OTAA, extracted DevAddr (watching out for the endianess), NwkSKey and AppSKey, reverted to all default parameters, so ADR as well and waited a bunch of uplinks on DR_5. 
After 15, I got one which was received!

pimente, 2021-10-02:
These launches will happen before the end of the year for sure 😄

MedadRufus, 2021-10-02:
For a flying balloon, its best to disable ADR. ADR is useful if the device is static. It reduces the power level/air time to as low as possible, while maintaining the link.

pimente, 2021-10-02:
Yes, ADR was disabled for my last flight in September. In fact I was using BlindADR which was a success until Turkey where it lost coverage. We should make more tests to understand the limits of this hack on the Helium network. Does is work well with ADR disabled? Do the credential expire?... But this sounds like a huge step forward to achieve circumnavigation 👍

MedadRufus, 2021-10-02:
I have always disabled ADR, and had no problem.

I am putting my thoughts and plan B for the expiring credentials issue here. Its somewhat seperate from the current issue: https://github.com/ImperialSpaceSociety/LoRaMac-node/issues/214

pimente, 2021-10-02:
Yep, we've been quite of course the initial issue in this thread 😄 
I'll make a new test now with ADR to try to discharge this problem.

MedadRufus, 2021-10-02:
I don't use ADR at all. I always use the highest possible power, and a fixed data rate. In the EU, I use DR_5 while in the rest of the world, I use DR_4. Note that in the US, the highest datarate is DR_4(not DR_5).

[3 more replies]